rdagger/micropython-ili9341 #1: Rotation Does Not Work Correctly

Pinnchus, 2020-12-13:
Using rotation 90 or 270, does not maintain across the code the correct new Height and Width, not allowing to use full display. 

Regards,
Pinnchus

rdagger, 2020-12-13:
Please give an example and explain where it fails.

Pinnchus, 2020-12-14:
If you initialize the screen with rotation=90 or rotation=270  using 
display = Display(spi, dc=Pin(4), cs=Pin(16), rst=Pin(17), rotation=**XXX**) 

when call display.clear(), will only fill 70% of the screen (approximately) and when you try to draw a line or a box, with the "new" height and width dump an error related to exceeding height.

Thanks for you time, and fast answer!!!!!!

rdagger, 2020-12-16:
When you initialize the display with rotation=90 or 270 you must specify the rotated width and height.  For example:
`display = Display(spi, dc=Pin(4), cs=Pin(16), rst=Pin(17), width=320, height=240, rotation=90)`

Otherwise the default width and height will be 240 and 320 respectively which will be reversed.

closed 2020-12-18 by rdagger

Pinnchus, 2020-12-18:
Thanks, do not  know that!!!