rdagger/micropython-ili9341 #3: 'machine' Module not found

metehus, 2021-02-21:
Hi, i'm trying to run the `demo_shapes.py` with an Raspberry PI Zero W, but everytime i try to run, it give me this error:
```
Traceback (most recent call last):
  File "demo_shapes.py", line 4, in <module>
    from machine import Pin, SPI
ModuleNotFoundError: No module named 'machine'
```

I already tried to install using `pip3 install machine`, but no success. What can i do?

rdagger, 2021-02-21:
The library is for MicroPython.

metehus, 2021-02-21:
> The library is for MicroPython.

Searched here and looks like it doesn't work with Raspberry PI, but thanks anyway

closed 2021-02-21 by metehus

gardinerwally, 2025-06-08:
I got it! YES! I loaded in test code to make the onboard LED blink. DID NOT WORK!  Changed the Interpreter type-  Tools>Option>Interpreter. I’m using MicroPythonRp2040 instead of the one at the top of the top of the list! Put the code in and keep changing the Interpreter setting until it works