rdagger/micropython-ili9341 #6: draw_vline boundary check error

CableGuy67, 2021-04-22:
Hello and thanks for this project. I'm using it on my Pi Pico and it is way faster than the others I tried. I think I found a small error where the bottom edge of the screen is detected incorrectly since it calls is_off_grid without subtracting one from the line height.

It is on line 577 and should read,

`        if self.is_off_grid(x, y, x, y + h - 1):
`
Then it matches the block call following like hline does which was how I found it since the horizontal lines of the rectangle were drawing just fine at the bottom of the screen.

rdagger, 2021-04-22:
Good catch!

rdagger, 2021-04-22:
I fixed the bug.  I also found a bug with the contrast code.  I don't have any ILI9341 boards anymore so please test the  _draw_vline_ and _contrast_ methods and let let me know if they work.  Thanks.

CableGuy67, 2021-04-23:
Thanks for the quick commit!
I see what you mean about the contrast thing but when I played a little with it I didn't see any effect. I wonder if the register that 0x53 loads the info for the Write CTRL Display stuff. It all defaults to 0x00 which seems to not be correct.
Also, it could be my display. It's a Waveshare 2.4 inch that does have a backlight pin (which I had wired to a pot) so it seems that the leds should be able to be controlled. It is possible I just didn't try the right combination of 0x51 and 0x53. Or, that my brian wanted to explode from reading the datasheet.
I'll play more with it this weekend and maybe see if I have any other ili9341 displays in the box. I'll let you know if I find anything of value. This can be closed though as the fix does seem to be correct (for both).

rdagger, 2021-04-24:
It looks like brightness control is hardware dependent.  As far as I can tell most of the ILI9341 modules have the backlight LED hardwired precluding software control.  Therefore, I removed the contrast method but left the constants in case someone has a supported board. 

You should be able to control the backlight brightness using PWM.  Some ILI9341 modules provide a transistor to facilitate brightness control.  Otherwise, you would need to supply and external transistor.

closed 2021-04-24 by rdagger