rdagger/micropython-ili9341 #8: Fonts Errors

thycoding, 2021-09-09:
Some of the fonts like IBMPlexMono12x24.c, UbuntuMono12x24.c and UnispaceExt12x24.c have errors and cannot be used. Please advise. Thank you.

rdagger, 2021-09-09:
Those 3 fonts have extended character sets so make sure you specify the correct _letter_count_ parameter.  Most of the fonts start at ASCII 32 and go to ASCII 127 so they have 96 characters.  The 3 aforementioned fonts extend to ASCII 255 so you would need to specify 224 for the _letter_count_ when you initialize the font.

Those 3 fonts are also quite large at 54 -56 KB.  You could have memory issues depending on your board's specs and how much you've put on it.  You can edit the font files to make them smaller.  For example you could delete the last 128 lines if you don't need the extended ASCII codes.

If you still have errors, please post the full error message and I will try to replicate the issue.

thycoding, 2021-09-09:
It's working now. Thanks for the clear and succinct explanation. Much appreciated.

> Those 3 fonts have extended character sets so make sure you specify the correct _letter_count_ parameter. Most of the fonts start at ASCII 32 and go to ASCII 127 so they have 96 characters. The 3 [...]

thycoding, 2021-09-09:
> Those 3 fonts have extended character sets so make sure you specify the correct _letter_count_ parameter. Most of the fonts start at ASCII 32 and go to ASCII 127 so they have 96 characters. The 3 [...]

It's working now. Thanks for the clear and succinct explanation. Much appreciated.

closed 2021-09-09 by thycoding