rdagger/micropython-ili9341 #10: Backlight not turning on

peterbmarks, 2022-08-31:
The library is working but my backlight remains off. It's not a hardware fault as the demo software worked.
The board I have is "ESP32 Arduino LVGL WIFI&Bluetooth Development Board 2.8" 240*320 Smart Display Screen 2.8inch LCD TFT Module With Touch WROOM" purchased from [AliExpress](https://www.aliexpress.com/item/1005004502250619.html).

I've tried setting each of the GPIO pins high and low in case that's how it's done. Any tips would be most welcome.

rdagger, 2022-09-01:
Some ILI9341 boards have the back light pin connected to a transistor or MOSFET so it can be controlled from a GPIO pin.  This is convenient if you want to turn the back light LED on and off from code such as if you wanted to implement a sleep mode.  My board didn’t come with the transistor, so I just wired it directly to 3.3 V.  Based on the schematic that you linked, it looks like your back light is driven by an N-Channel MOSFET on GPIO 21.  You just need to toggle GPIO 21 to turn the back light on or off.

peterbmarks, 2022-09-01:
Wow! Thank you so much. I now have bright bouncing boxes. I had tried a range of GPIO pins but gave up before 21. You are a legend.

closed 2022-09-01 by peterbmarks