rdagger/micropython-ili9341 #12: Moving the Available Space

jouellnyc, 2022-11-10:
Howdy. Really great Library! Already used it easily.

I am using the ili9341 with rotation=90

I see that there's about a 1/2 inch of usable space on the "left"  (i.e 'bottom' of the ili9341 with rotation=0)

[Pic](https://ibb.co/Y3GZ0rD)

I have tried to manipulate VSCRDEF but it seems like I just end up squish the text all together.

Any ideas on how to achieve filling up the full screen using the rotation=90?

rdagger, 2022-11-10:
I can’t tell from your picture what’s going on.  That may just be normal for your particular display.  If you search Google Images for Ili9341, there are several variants that have unequal borders.

Assuming your display is 240 x 320, I would draw a single dot at each corner (0, 0), (239, 0), (0, 319), (239, 319). If the driver is working, you will see all 4 pixels lit. If not, then I would start moving the missing dots away from the corner coordinates to see how many pixels are missing. Then let me know how many pixels the display is off.

I haven’t seen this problem with the Ili9341 but I have seen somewhat similar problems on other displays because of variations between revisions and vendors.  Do you have a link to the display you purchased?

I would also try with another driver such as the [CircuitPython Display IO Driver](https://github.com/adafruit/Adafruit_CircuitPython_ILI9341) and see if it does the same thing.

jouellnyc, 2022-11-10:
Thanks for very much for the reply.

Yes it's a 2.8 In 320x240 [Display from Hosyond](https://www.amazon.com/dp/B09XHJ9KRX?ref=ppx_yo2ov_dt_b_product_details&th=1), gotten from AMZN. 

I also share your thoughts/concerns about the Hardware. It really seems like a lot of revisions/vendors/etc...

- I tried your pixel tests (using rotation=90) (320 horizontal - 240 vertical):

```
>>> display.draw_pixel(0,0,white)
>>> display.draw_pixel(0,239,white)
>>> display.draw_pixel(239,0,white)
>>> display.draw_pixel(239,239,white)

```
I see all 4 pixels. Here's where those commands put the pixels (with obvious large dot mark ups from me):

[Pic](https://ibb.co/QQtcVc9)

319 was too big of a value:

```
>>> display.draw_pixel(239,319,white)
y-coordinate: 319 above maximum of 239.
>>> display.draw_pixel(0,319,white)
y-coordinate: 319 above maximum of 239.

```

I was able to put a pixel in every corner using:

```
>>> display.draw_pixel(319,239,white)
>>> display.draw_pixel(319,0,white)
>>> display.draw_pixel(0,0,white)
>>> display.draw_pixel(0,239,white)

```
[Pic](https://ibb.co/Kz5q5Jf)

- Interestingly,  313 is the biggest value I can specify to draw text and the errors seems to refer to different number than I specify. I.E this does print to the screen:

```
display.draw_text(313,0, 'test',sm_font, white)
x-coordinate: 320 above maximum of 319.
x-coordinate: 332 above maximum of 319.
Invalid width 0 or height 0
```

I only see the 't' and 'ext' is off the screen, but maybe that is meaningful?

rdagger, 2022-11-10:
It's been a while since I looked at this code.  I think you are reversing the height and width.  Try the default parameters of width=240, height=320, rotation=0.  Also make sure you are using the correct limits.  I think you have a typo in your above message with pixels (239, 239) and (0,239).  Also try running the demos included with the repo and let me know if they work.

rdagger, 2022-11-10:
Definitely take a look at and try: https://github.com/rdagger/micropython-ili9341/blob/master/demo_orientation.py

jouellnyc, 2022-11-11:
Thanks very much for that. Sorry about the typo. Maybe some foo came across from my testing. I edited the comment.

I ran the demo - still the little area persists off to the side. Maybe it's a weird hardware thing. 

I'll try the Adafruit lib and maybe see if I can post to a hardware forum and compare model numbers. 

Thanks for your time in looking and of course the great driver.

rdagger, 2022-11-11:
Many of the displays have a larger border on the bottom presumably to have permanent silk screened touch buttons.  Your border doesn't look much different from the vendor's photo on Amazon.  Please let me know if the other libraries give you additional fill and I will compare the settings.

[image]

closed 2022-11-11 by rdagger