rdagger/micropython-ili9341 #13: Q: Support for ESP S3 BOX?

jouellnyc, 2023-03-03:
Howdy. I noticed the esp32_s3_box has an ili9342c based display:

https://github.com/espressif/esp-box/blob/master/docs/hardware_overview/esp32_s3_box/hardware_overview_for_box.md

Clearly, that is different from an ili9341 and  was not intended for that usage.

But, being curious, I tried. 

I can use this repo and write to the screen but it comes out as  a mirror image and write down the screen instead of across. 
What I am uncertain of is there would be support for the ili9342c/esp32_s3_box?

I've tried other repos w/o success using the esp32_s3_box  and micropyton, so figured I'd toss the Q out there. 

Thanks in advance!

rdagger, 2023-03-03:
I don't have an ili9342c.  I took a quick look at the datasheet and it does look similar.  The issues you're having are probably due to differences in the MADCTL register (memory access control).  You may be able to just modify the ROTATE constants.  Currently my library uses:
```
    ROTATE = {
        0: 0x88,
        90: 0xE8,
        180: 0x48,
        270: 0x28
    }
```

I'd really need the display to know for sure but the following values may work:
<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 14">
<link id=Main-File rel=Main-File
href="file:///C:\Users\REGIST~1\AppData\Local\Temp\msohtmlclip1\01\clip.htm">
<link rel=File-List
href="file:///C:\Users\REGIST~1\AppData\Local\Temp\msohtmlclip1\01\clip_filelist.xml">

</head>

<body link=blue vlink=purple>

Rotation | Hex | Binary
-- | -- | --
Portrait | 0x08 | 0b0000 1000
Landscape | 0x68 | 0b0110 1000
Inverted   Portrait | 0xC8 | 0b1100 1000
Inverted   Landscape | 0xA8 | 0b1010 1000
Portrait   Mirrored | 0x48 | 0b0100 1000
Landscape   Mirrored | 0x28 | 0b0010 1000
Inverted   Portrait Mirrored | 0x88 | 0b1000 1000
Inverted   Landscape Mirrored | 0xE8 | 0b1110 1000

</body>

</html>

So the new constants would be:
```
    ROTATE = {
        0: 0x08,
        90: 0x68,
        180: 0x88,
        270: 0xE8
    }
```

There could be other changes too.  Again, it's very difficult without having a display to test.

jouellnyc, 2023-03-03:
Oh Wow!. That's really super!

I played around with the values you suggested. If I set `rotation=0` and 

```
ROTATE = {
        0: 0xC8,
        90: 0xA8,
        180: 0x48,
        270: 0x28
    }

```    

and also: 

```
width=320
height=240

```

everything works perfectly, really. I can read from left to right, the screen is filled perfectly and colors are crisp!

Not to exclude this fact, but  I can use the real RGB values for color565().
(For reference,  withe esp s3 lite box, they were basically inverted)

So, in summary, it works perfectly. Wow and thank you so much!

(My email is on my profile. If you were to send your address, I am only too happy to send you an S3 Box for you to use/test)

rdagger, 2023-03-03:
Thanks, I appreciate the offer! Unfortunately, I'm not able to take on any new projects right now.

closed 2023-03-03 by rdagger

jouellnyc, 2023-03-04:
Sure thing, I should have explicitly said there are no expectations. I.E it was purely a thank you and offer stands.
(I had been at road block on my project and now there is a green light again).  Many thanks and virtual high five!!