rdagger/micropython-ili9341 #15: Performance mods

sumnerk, 2023-05-07:
I've been tinkering with your driver to allow me to drive an ILI9488 and along the way found some nice (I think, but I'm new to all this python and microcontroller stuff) ways to enhance the performance of your driver for the ILI9341 which I also use and find to be excellent !
I concentrated on the block() function as it's the workhorse and pre-allocated some bytearrays to hold the page and column coordinates as well bytearrays for the command bytes themselves. I also moved away from calls to write_cmd() and write_data() and instead switched to direct spi.write() calls
Some numbers
Filled / unfilled rectangle drawing - before 2330ms, after 1688
Filled circles (192) - before 6451ms, after 2223
Filled squares (192) - before 456, after 323

Fairly decent improvements

I'm not to familiar with how GitHub works but I think I can attach the code to this ticket so you can have a look and decide if it's worth including in your driver
[ili9341.txt](https://github.com/rdagger/micropython-ili9341/files/11415910/ili9341.txt)

rdagger, 2023-05-07:
Thanks I appreciate the feedback.  I'll take a look at your code.  Are you using MicroPython or CircuitPython?

I have been experimenting with ways to speed up SPI for a new high speed ADC library.  However, I don't normally have time to go back and update older libraries unless there's interest.

sumnerk, 2023-05-07:
I'm using micropython on pico and pico w for a hue light controller and using the display to provide feedback.
Just playing around really and trying to learn python and disappeared down the performance tuning rabbit hole for a day!

Am currently struggling with use of spi.read() to allow me to receive info from the LCD in response to sending various status commands (e.g. RDDID, RDDST etc) but I'm not having too much luck.

rdagger, 2023-11-22:
I replaced ustruct with your bitwise logic and observed a performance improvement. However, I chose to keep the SPI writing methods to maintain compatibility with both MicroPython and CircuitPython. This decision was also made to avoid introducing breaking changes and to prioritize code readability and maintainability.

closed 2023-12-05 by rdagger