rdagger/micropython-ili9341 #17: Display Mirrored

2Bor3d, 2023-11-02:
I am using a Raspberry Pi Pico W with a 128x160 pixel LCD display from AZ-Delivery.
In most cases, the mirrored display isn't a problem, because I can just code everything mirrored.
But as far as I can see, there is no way to do this, when trying to display text. 
Am I doing something wrong? Is this a bug? Is there a way to just mirror the display again so that it is correct?

rdagger, 2023-11-03:
I'm not sure what you mean by mirror.  Do you mean like text in a mirror such as redruM in the Shining?

I think you are probably referring to the lack of 180 and 270 degree rotation.  I went ahead and added a rotate_180 parameter to the draw_text method.  It can be used in conjunction with the landscape parameter to rotate text in the 4 cardinal directions.

I created a demo called demo_fonts_rotated.py so you can see how it works.

It would be better to replace the landscape and rotate_180 with a single rotation parameter but I don't want to create breaking changes.

Here's the code if you're interested.  It's more of a hack:

```
if rotate_180:
    # Manually rotate the buffer by 180 degrees
    # ensure bytes pairs for each pixel retain color565
    new_buf = bytearray(len(buf))
    num_pixels = len(buf) // 2
    for i in range(num_pixels):
        # The index for the new buffer's byte pair
        new_idx = (num_pixels - 1 - i) * 2
        # The index for the original buffer's byte pair
        old_idx = i * 2
        # Swap the pixels
        new_buf[new_idx], new_buf[new_idx + 1] = buf[old_idx], buf[old_idx + 1]
    buf = new_buf
```

If you really want mirrored text (as in how text looks in a mirror) then you could use a similar approach to above.  To mirror the text horizontally, you would need to reverse the order of the pixels in each row but not the rows themselves.  Here's a theoretical example that I have not tested:

```
def mirror_buffer_horizontally(buf, width, height):
    # Create a new buffer for the mirrored image
    new_buf = bytearray(len(buf))
    
    for row in range(height):
        for col in range(width):
            # The index for the original buffer's byte pair
            src_idx = (row * width + col) * 2
            # The index for the mirrored buffer's byte pair
            # It starts at the end of the row and moves backwards
            mirror_idx = (row * width + (width - col - 1)) * 2
            # Copy the pixel pair to the mirrored position
            new_buf[mirror_idx], new_buf[mirror_idx + 1] = buf[src_idx], buf[src_idx + 1]
    return new_buf
```

2Bor3d, 2023-11-05:
Thanks for the fast reply!
I meant that the text was mirrored, as in how text looks in a mirror. I tested your code and it works for me. The only problem is that the text is on the wrong side, but that is something I can address myself.

closed 2023-11-05 by 2Bor3d