rdagger/micropython-ili9341 #20: Anyone got the Lilygo T4 working ? https://www.lilygo.cc/products/t4

clivebearmanhub, 2024-01-16:
Hi folks,

Firstly, this library is fantastic, and thanks so much for developing it. I've got it working on a standard ESP 32 dev board and separate display, but just can't get it working on the Lilygo T4 https://www.lilygo.cc/products/t4. Nothing displays at all. Sorry for posting if this is not the right forum.

rdagger, 2024-01-17:
Please post your code.  Do you have v12 or 13?

clivebearmanhub, 2024-01-22:
Thanks for responding. I eventually got it to work by removing the baudrate=40000000 parameter from the SPI creation. I also used the pin mapping according to LillyGo's T4 v13 table here https://github.com/Xinyuan-LilyGO/LilyGo_Txx . Using your "demo_images.py" as an example, the modified example now reads...

from time import sleep
from ili9341 import Display
from machine import Pin, SPI

def test():
    spi = SPI(1, sck=Pin(18), mosi=Pin(23), miso=Pin(12))
    display = Display(spi, dc=Pin(32), cs=Pin(27), rst=Pin(5))
    display.draw_image('images/RaspberryPiWB128x128.raw', 0, 0, 128, 128)
    sleep(2)
etc...

Thanks once again.

closed 2024-01-22 by rdagger