rdagger/micropython-ili9341 #22: Can't write text

forcie, 2024-02-13:
Hi,

I don't know what I'm doing wrong, but whenever I try to write text, it's either all mirrored, or the font itself seems messed up.

I get the same result in real life as what you can see in this simulation:
[](https://wokwi.com/projects/389429329585761281)

Thanks for any help possible.

rdagger, 2024-02-14:
Do all the other demos besides fonts work okay?
What specific ILI9341 board do you have?

forcie, 2024-02-14:
I've got this ILI9341 2.8" inch TFT w/ touch at 240x320, which seems pretty standard, and I've seen it on your website:
[https://ibb.co/G7BQ1ZS](https://ibb.co/G7BQ1ZS)
As for the other demos, they seem to work just fine, although for many of them it's not a definitive answer as the effect that I get with draw_text I would describe as "the viewer is watching from _behind_ the display", so for a color palette the result is the same.
That being said, demo_fonts8x8.py seems to give the expected results. But all demos using draw_text lead to weird stuff:
[https://ibb.co/RQf7nKp](https://ibb.co/RQf7nKp)

rdagger, 2024-02-14:
I'm guessing you have a variant of the ILI9341.  Please provide a link to where you bought it.
What version of MicroPython are you running?

forcie, 2024-02-14:
https://www.aliexpress.us/item/3256806129218488.html
Granted the picture of the 2.8" shows v1.0, the one I received shows v1.1
But the funny thing is that Wokwi's emulator (link in my first message) gives the exact same result as my real life display, and I'd be surprised if that emulator was based on some kind of new variant of ILI9341.
As for my setup:
`MicroPython v1.22.1 on 2024-01-05; Raspberry Pi Pico W with RP2040`

rdagger, 2024-02-14:
Here's some code to draw a letter F with lines.  Is the orientation correct?  Are the colors for each line correct?
```
    # Define colors
    COLOR_RED = color565(255, 0, 0)
    COLOR_BLUE = color565(0, 0, 255)
    COLOR_GREEN = color565(0, 255, 0)
    COLOR_YELLOW = color565(255, 255, 0)
    COLOR_PURPLE = color565(128, 0, 128)
    COLOR_CYAN = color565(0, 255, 255)
    COLOR_MAGENTA = color565(255, 0, 255)
    COLOR_ORANGE = color565(255, 165, 0)
    COLOR_WHITE = color565(255, 255, 255)

    # Outer Vertical Line
    display.draw_line(41, 21, 41, 299, COLOR_RED)

    # Inner Vertical Line
    display.draw_line(61, 41, 61, 299, COLOR_BLUE)

    # Outer Top Horizontal Line
    display.draw_line(41, 21, 181, 21, COLOR_GREEN)

    # Inner Top Horizontal Line
    display.draw_line(61, 41, 181, 41, COLOR_YELLOW)

    # Outer Middle Horizontal Line
    display.draw_line(61, 160, 161, 160, COLOR_PURPLE)

    # Inner Middle Horizontal Line
    display.draw_line(61, 141, 161, 141, COLOR_CYAN)

    # End Cap on Outer Vertical Line (Bottom)
    display.draw_line(41, 299, 61, 299, COLOR_MAGENTA)

    # End Cap on Outer Top Horizontal Line (Right)
    display.draw_line(181, 21, 181, 41, COLOR_ORANGE)

    # End Caps on Inner Lines (Middle)
    display.draw_line(162, 141, 162, 160, COLOR_WHITE)
```

forcie, 2024-02-14:
Yes! All colors & lines match their descriptions (besides the fact that I've been looking at the display upside down)
[image]

rdagger, 2024-02-15:
I still think you have a variant that handles memory access control differently.  It's very hard to troubleshoot without having the display.   

I wiped an ESP32 and loaded the latest version of Micropython and the latest commit from my repo.  I tested all the font demos, and they work as expected.

Here's a hack that might work.  Add the code below to the _draw_letter_ method of ili9341.py.  Place the code after the line:
`buf, w, h = font.get_letter(letter, color, background, landscape)`

Here's the new code:
```
        # Initialize a new buffer
        new_buf = bytearray(len(buf))
        if landscape:
            # Perform combined mirroring operation
            for row in range(h):
                for col in range(w):
                    # Calculate the index for the pixel in the original buffer
                    original_index = (row * w + col) * 2  # 2 bytes per pixel for color565 format
                    # Calculate the new index for the pixel in the mirrored position
                    # Get the pixel from the opposite corner
                    mirrored_index = ((h - 1 - row) * w + (w - 1 - col)) * 2
                    # Copy the pixel data to the new buffer at the mirrored position
                    new_buf[mirrored_index], new_buf[mirrored_index + 1] = buf[original_index], buf[original_index + 1]
        else:
            for col in range(h):
                for row in range(w):
                    # Calculate the index for the pixel in the original buffer
                    old_index = (row * h + col) * 2
                    # Directly map original rows to new columns without inverting their order.
                    new_row = col  # Correct for 90-degree rotation (transposition part)
                    new_col = row  # Directly use original_row, removing the mirroring adjustment
                    # Calculate the new index in the buffer for the rotated position
                    # This adjustment assumes the width and height are effectively swapped due to rotation
                    new_index = (new_row * w + new_col) * 2 
                    # Copy the pixel data to the new buffer at the rotated position
                    new_buf[new_index], new_buf[new_index + 1] = buf[old_index], buf[old_index + 1]

        # Update the original buffer with the mirrored image data
        buf = new_buf
```

forcie, 2024-02-16:
Genius, that worked immediately. Thank you so much for your help and this fix.
If I indeed have a variant of ILI9341, wokwi.com's emulator must be implemented according to that variant as well, because I am at work without the real display and this fix worked on the emulator. Can't wait to try that at home!

rdagger, 2024-02-16:
I thought more about your issue and realized it could not be a variant issue because _draw_text8x8_ writes to memory in the same way as _draw_text_.  Since you reported that the 8x8 font works, there is nothing that would cause this behavior other than if the xglcd_font.py was modified.  

I took a look at your wokwi app and found that xglcd_font.py is not the correct version for this repo.  I also checked ili9341.py and it is the correct and latest version, so I don't know how you got the wrong xglcd_font.py.  It is either from a very old revision or from another one of my display driver projects.

Please clone the latest version from my repo and see if it works.

forcie, 2024-02-16:
My sincere apologies, I really don't understand how this happened. I still get unexpected results with all latest files together but I will investigate further before I bother you again.

closed 2024-02-16 by rdagger