rdagger/micropython-ili9341 #26: FrameBuffer

TamTamPy, 2024-03-25:
Hey,

i know this is quite old code. But i was wondering... could I use the framebuf library with this?
Specifically: I want to have a "start-up-animation" like every gaming console has. But for the framebuffer i would need bitmap images.
Your library needs the raw rgb565... so I don't know where to go from here.

Maybe a little help for some micro-controlling-newbie?

TamTam

rdagger, 2024-03-25:
The MicroPython framebuf library supports RGB565:
https://docs.micropython.org/en/latest/library/framebuf.html#framebuf.framebuf.RGB565

You could use a sprite sheet.  There's an example on my website:
https://www.rototron.info/projects/esp32-pwned-password-checker/#sprites

rdagger, 2024-03-25:
I added an animation demo (demo_animated_sprite.py).

https://github.com/rdagger/micropython-ili9341/assets/106355/a5d91833-15cb-4cf5-a172-1b2302ce1ce8

TamTamPy, 2024-03-25:
okay wow... this is exactly what I need!  :):):)
thank you so much !

TamTamPy, 2024-03-26:
https://github.com/rdagger/micropython-ili9341/assets/152076757/fa4ae275-7df5-4059-a8be-69a825e324d0

would u have an idea why the animation is so slow on my setup?

ESP32-S3  (Arduino Nano Version)
Micropython (of course)

closed 2024-03-30 by rdagger