rdagger/micropython-ili9341 #27: How to adapt this library to the Raspberry pi pico

MohamedZaky0, 2024-03-26:
Hello,
I would like to use this library on Raspberry pi pico but as i have noticed in the tutorial it is for ESP32 and the pinout also for ESP32, Could you guide me for any documentation or anything so i can adapt it for the Raspberry pi which uses micropython?

rdagger, 2024-03-26:
I don't have one available to test but you should be able to do something like this:
```
spi = SPI(0, baudrate=40000000, sck=Pin(2), mosi=Pin(3))
display = Display(spi, dc=Pin(6), cs=Pin(5), rst=Pin(7))
```

You can try lowering the baud rate if you have issues. 
 
You will be limited in terms of fonts and images because the pico has much less SRAM them the ESP32.  You get a lot more flexibility on boards with lots of SRAM or PSRAM.

MohamedZaky0, 2024-03-26:
Is there any limitation for the pins ? is there any specific pins that i should connect the IFI to the Rasp? I am also using the touchscreen

rdagger, 2024-03-26:
Take a look at the pinout on the [Raspberry Pi Pico page](https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html).
I would stick to pink pins labelled SPI (GPIO 0-19).  Please note some are SPI bus 0 and some are SPI bus 1.  You might as well use the default bus  on SPI0 (GPIO 16-19) for the display and any of the available SPI1 pins (GPIO 8-15) for the touchscreen.  I have found it is better to run them on separate buses because of the difference in speed.

rdagger, 2024-03-28:
I verified that the following pins work on a pico:
```
spi = SPI(0, baudrate=40000000, sck=Pin(18), mosi=Pin(19))
display = Display(spi, dc=Pin(4), cs=Pin(17), rst=Pin(5))
```

closed 2024-03-30 by rdagger