rdagger/micropython-ili9341 #29: Invertid colors: BRG instead of RGB

IvanBella, 2024-03-28:
Hey, sorry to bother you with this but I was wondering if you could help me with this because I can’t find a way to fix it. On my screen (a ROHS 2.4” TFT SPI 240x320 V1.3) the colors appear inverted. 255,0,0 for example comes out as blue. Do you have any idea what it might be and how I could fix it?
Thank you so much for sharing your great work man!

rdagger, 2024-03-28:
If every color is inverted then red would be cyan, green would be magenta and blue would be yellow.  
[image]

In this case you could try Display Inversion ON (see datasheet 8.2.16):
`display.write_cmd(display.INVON)`

To disable inversion, use Display Inversion OFF:
`display.write_cmd(display.INVOFF)`

IvanBella, 2024-03-28:
Dear me! That’s a quick response! Thanks!
I think the problem I have is that for some reason what should be red (255,0,0) comes out as blue and vice versa.
If I turn INV on, as you said, RGB becomes MCY, but as my display shows BGR, I get YCM. Scratching my head…

[image]
[image]

IvanBella, 2024-03-28:
Sorry, meant to say: if I turn INV on, as you said, RGB becomes CMY, but as my display shows BGR, I get YMC. Scratching my head…

IvanBella, 2024-03-28:
Found this, that maybe can help but would know how to implement it…

[image]

IvanBella, 2024-03-29:
Found it!
The RGB value in MADCTL was set to 1. 
so for example when rotation was set to 0 (0x88), the binary value was 1000100.
I have set this to (0x80), which gives me 1000000. Now colors are shown as RGB and not BGR :)
[image]

IvanBella, 2024-03-29:
The problem I need to solve now is why pics seem to be shown in negative…
[image]

IvanBella, 2024-03-29:
> Try the following command after you initialize the display:
> 
> `display.write_cmd(display.MADCTL, display.rotation & 0b11110111)`

Sorry, browser didn’t refresh and I did not see your message. I’ll try it out and see what results I get. Thanks!

IvanBella, 2024-03-29:
Do you think the negative picture problem has to do with gamma value?

rdagger, 2024-03-29:
Please post a picture of the negative picture problem when running demo_images.py

IvanBella, 2024-03-29:
[image]

IvanBella, 2024-03-29:
I hope I uploaded it properly this time. 
Thank you for looking into this!

rdagger, 2024-03-29:
The colors in the image are not inverted; they appear to be washed out. This issue is likely due to the positive and negative gamma correction settings. It seems you have a variant of the ILI9341 display. Your variant processes RGB/BGR and gamma corrections differently from mine, and there might be other differences as well.

Unfortunately, I cannot adjust the gamma settings without access to your specific display variant. However, I found another set of gamma correction values online. Try replacing the following two lines in the initialization method of ili9341.py:

```
self.write_cmd(self.GMCTRP1, 0x0F, 0x31, 0x2B, 0x0C, 0x0E, 0x08, 0x4E,
	0xF1, 0x37, 0x07, 0x10, 0x03, 0x0E, 0x09, 0x00)
self.write_cmd(self.GMCTRN1, 0x00, 0x0E, 0x14, 0x03, 0x11, 0x07, 0x31,
	0xC1, 0x48, 0x08, 0x0F, 0x0C, 0x31, 0x36, 0x0F)
```
        
Here's the replacement code:

```
self.write_cmd(self.GMCTRP1, 0x0F, 0x2A, 0x28, 0x08, 0x0E, 0x08, 0x54,
	0xA9, 0x43, 0x0A)
self.write_cmd(self.GMCTRN1, 0x00, 0x15, 0x17, 0x07, 0x11, 0x06, 0x2B,
	0x56, 0x3C, 0x05, 0x10, 0x0F, 0x3F, 0x3F, 0x0F)
```

Alternatively, you could try commenting out these two lines to see how the default settings perform. Be sure to power cycle the board to ensure the defaults are applied.

IvanBella, 2024-03-29:
You’re a star! 
Commenting out did the trick. Images show a little blueish but I can work with that and try to play with the parameters. 
Thank you so much for your help!

rdagger, 2024-03-29:
I added 2 optional parameters to the ili9341 constructor:

- bgr (Optional bool): Swaps red and blue colors (default True)
- gamma (Optional bool): Custom gamma correction (default True)

With your board, you would set both to false:
`display = Display(spi, dc=Pin(4), cs=Pin(16), rst=Pin(17), bgr=False, gamma=False)`

Please let me know if the changes work.

IvanBella, 2024-03-31:
That is wonderful!! Thank you so much my friend!

closed 2024-04-02 by rdagger