rdagger/micropython-ili9341 #31: External SD Card help

Mark25A, 2024-05-17:
Anyone knows how I can use this with an external SD Card?

I need it for my school project and it's the only thing that doesn't work. I want to simply load an image from the micro-sd card (not the tft's) when touching the screen at a specified coordinate. I can load it from the esp32 no problem, but the moment I initialize the sd card, the screen turns black.

I use micropython, ESP32 and an sd card library: https://github.com/micropython/micropython-lib/blob/master/micropython/drivers/storage/sdcard/sdcard.py

If I don't initialize it, it works fine.

Anyone knows something? I tried controlling high and low, even disconnecting the sd card and it would still give me a black screen.

rdagger, 2024-05-17:
Thank you for your interest in our project! We appreciate your enthusiasm and willingness to engage. However, we kindly request that the GitHub Issues section be reserved for reporting bugs, requesting features, or discussing specific problems related to the repository.

For general questions or broader discussions, please consider using the MicroPython Team GitHub Discussions:  
https://github.com/orgs/micropython/discussions

Your question has to do with conflicts when running more than 2 SPI devices.  The MicroPython SD card class is a powerful tool, but it may not always integrate smoothly with other SPI devices. This is due to potential conflicts and timing issues that can arise when multiple devices share the same SPI bus.

When I get some free time, I will take a look but your best bet to get a quick response is to post your question to the aforementioned discussion site.

Mark25A, 2024-05-17:
I apologize, I did not mean to break the rules.

Thank you for the reply, I will try it there.

And thank you for this amazing library, I was able to do even more than what I wanted for my painting app project, it helped me out so much!

rdagger, 2024-05-18:
I tested the library with an SD card and it works.  I created a demo called demo_sdcard.py

Mark25A, 2024-05-18:
> I tested the library with an SD card and it works. I created a demo called demo_sdcard.py

Thank you, you're a genius. I can't test it right now because turns out the sd card is faulty, no longer recognizable on any device. 

But I see you used an interrupt? I thought to use it and maybe only then initialize the sd card. Either way, thank you for your time!

Mark25A, 2024-05-18:
Thanks again, I was able to rewrite the test code for myself with the help of initializing the sd card using an interrupt, now it works. I do have a small issue with the display filling a small portion of the screen with the color white when I do that, but I'll figure it out. I appreciate you so much!

closed 2024-05-18 by Mark25A