rdagger/micropython-ili9341 #32: Mirrored text and touch acting weird

loreblb, 2024-06-20:
Hi,

I'm trying to perform a couple of simple actions using this library on a Raspberry Pi Pico running MicroPython, but I've encountered some issues:

1. The text is mirrored. I'm not talking about the orientation; the entire line is mirrored. Additionally, it is slightly outside of the margin, appearing to be offset by 1px on both the X and Y axes.
2. The touch functionality is acting weird. The X and Y values I get from the touch function look random (?). Sometimes, the circle gets "cut" as if the display is vertically separated into two parts.

See the video:

https://github.com/rdagger/micropython-ili9341/assets/99609110/40f83d89-141a-4121-a339-489ded6eb093

This is the code I'm running:

```
from ili9341 import Display, color565
from xpt2046 import Touch
from machine import Pin, SPI
from xglcd_font import XglcdFont

touch = None

def touch_handler(x, y):
    global touch
    
    display.draw_circle(x, y, 3, color565(255, 255, 255))
    
    print(f"Touch at ({x}, {y})")

spi = SPI(0, baudrate=51200000, sck=Pin(18), mosi=Pin(19))

cs_pin = Pin(17, Pin.OUT)
dc_pin = Pin(16, Pin.OUT)
reset_pin = Pin(20, Pin.OUT)
display = Display(spi, cs=cs_pin, dc=dc_pin, rst=reset_pin, rotation = 0, bgr=False, width=240, height=320)
display.clear()
display.display_on()

unispace = XglcdFont('fonts/Robotron7x11.c', 7, 11)

display.draw_text8x8(0, 0, "Hello World", color565(255, 255, 255))
display.draw_hline(10, 10, 230, color565(255, 255, 255))

spi1 = SPI(1, baudrate=1000000, sck=Pin(14), mosi=Pin(11), miso=Pin(12))

touch_cs_pin = Pin(13, Pin.OUT)
touch_int_pin = Pin(9, Pin.IN)
touch = Touch(spi1, cs=touch_cs_pin, int_pin=touch_int_pin, int_handler=touch_handler)

while True:
    pass
```

rdagger, 2024-06-20:
There are variants of the ILI9341.  The orientation of the touchscreen is not always the same.  You may need to reverse the width and height parameters when you initialize the touchscreen.  

You may also need to swap the x and y values in the touchscreen_press() callback and/or flip the x and/or y values as show here:
x, y = y, x  # Some screens require swapping X and Y
x = (self.display.width - 1) - x  # Some screens require flipping X
y = (self.display.height - 1) - y  # Some screens require flipping Y

You may have to try different combinations to get it working.

In terms of the font mirroring.  You are loading an XGLCD font but you are using the draw_text8x8 method which uses the built-in 8x8 Micropython font.  Does the mirroring also occur if you draw an XGLCD font like Robotron7x11?

The mirroring problem you described happened to another user when they used the wrong version of xglcd_font.py.  Only the most recent version of xglcd_font.py and ili9341.py from this repo should be used.

loreblb, 2024-06-21:
Hi, thanks for the quick reply!
I'll try to play around with touch coordinates and screen width.
In the code I also tried XGLCD fonts (Robotron7x11 too) but the issue was the same.
I downloaded the latest .py files from this repo yesterday so everyone of them should be up to date.
The mirroring issue also appears with the hline (see the video), it looks like al the pixels on the screen are horizontally mirrored (?).

rdagger, 2024-06-21:
Your variant probably requires a mirror option for the memory access control register.  That should not be too hard to implement.  I'll try to get something out tonight.

Where did you buy your display and what model ILI9341 is it?

rdagger, 2024-06-22:
I added a mirror option to the Display constructor.  I also added a demo called demo_mirror.py to show you how to use it.

loreblb, 2024-06-22:
I bought this display from Aliexpress, and there wasn't much info except for the driver type (ili9341).Tomorrow I'll try the code you updated and give you feedback. Thanks!

loreblb, 2024-06-23:
Ok, i'm back with some interesting news.
First of all the mirror option works correctly.

Then playing around with the display i found out that it is not 240x320 as stated in the product description, but a 128x160! It's not a big deal, but I would have preferred to know earlier.

So the touch, with the right size (and Y flipping too), is now working correctly.
But then i found another couple of problems:

- My screen needs an offset for X and Y, the first pixel is at x:2 and y:1, the last one is at x:129 and y:160 (could maybe be fixed by adding an optional offset parameter?).
- When drawing some text out of border, all the pixels of the letters are mixed up (shouldn't I get an error or shouldn't the text just be cut off?).

[image]

This is the code I'm running:
(Also, as stated in the code below, i commented out line 214 and 215 of ili9341.py, so maybe the problem of the mixed up text is happening because I'm simply trying to do something that I'm not allowed to do)

```
from ili9341 import Display, color565
from xpt2046 import Touch
from machine import Pin, SPI
from xglcd_font import XglcdFont

touch = None
display = None

def touch_handler(x, y):
    global touch
    global display
    
    y = (display.height - 1) - y
       
    display.draw_circle(x, y, 3, color565(255, 255, 255))
    
    print(f"Touch at ({x}, {y})")

spi = SPI(0, baudrate=51200000, sck=Pin(18), mosi=Pin(19))

cs_pin = Pin(17, Pin.OUT)
dc_pin = Pin(16, Pin.OUT)
reset_pin = Pin(20, Pin.OUT)

# Looks like Min-X is 2 and Min-Y is 1

# Because of the x and y offset, the screen can't be used correctly using these values
# display = Display(spi, cs=cs_pin, dc=dc_pin, rst=reset_pin, rotation=0, mirror=True, bgr=False, width=128, height=160)

# AssertionError: hlines must be a non-zero factor of height.
# Had to comment out line 214-215 of ili9341.py module to work with this custom size.
display = Display(spi, cs=cs_pin, dc=dc_pin, rst=reset_pin, rotation=0, mirror=True, bgr=False, width=130, height=161)

display.clear()
display.display_on()

# First pixel (top-left) is at X:2 Y:1
display.draw_pixel(2, 1, color565(255, 255, 255))
display.draw_text8x8(3, 4, "1", color565(255, 255, 255))

# Last pixel (bottom-right) is at X:129 Y:160
display.draw_pixel(129, 160, color565(255, 255, 255))
display.draw_text8x8(120, 150, "2", color565(255, 255, 255))

display.draw_text8x8(2, 40, "Mirror!", color565(255, 255, 255))

display.draw_text8x8(2, 40, "Mirror!", color565(255, 255, 255))
display.draw_text8x8(76, 100, "Border!", color565(255, 255, 255))
display.draw_text8x8(77, 110, "Border!", color565(255, 255, 255))
display.draw_text8x8(78, 120, "Border!", color565(255, 255, 255))

spi1 = SPI(1, baudrate=1000000, sck=Pin(14), mosi=Pin(11), miso=Pin(12))

touch_cs_pin = Pin(13, Pin.OUT)
touch_int_pin = Pin(9, Pin.IN)
touch = Touch(spi1, cs=touch_cs_pin, int_pin=touch_int_pin, int_handler=touch_handler, width=130, height=161, x_min=100, x_max=1962, y_min=100, y_max=1900)

while True:
    pass

```

rdagger, 2024-06-24:
I think you have an ST7735 display instead of an ILI9341. The initialization registers for these displays are different, albeit similar. I recommend you try a MicroPython driver specifically designed for the ST7735 to ensure proper initialization and functionality.

rdagger, 2024-06-24:
Here's the init sequence based on the Adafruit ST7735 library:

```
# Using ILI9341 constants where possible

# Reset and delay
self.write_cmd(self.SWRESET)
sleep(0.150)  # Delay 150ms

# Exit sleep mode and delay
self.write_cmd(self.SLPOUT)
sleep(0.255)  # Delay

# Frame rate control (In normal mode/full colors)
self.write_cmd(self.FRMCTR1, 0x01, 0x2C, 0x2D)

# Frame rate control (In idle mode/8 colors)
self.write_cmd(self.FRMCTR2, 0x01, 0x2C, 0x2D)

# Frame rate control (In partial mode/full colors)
self.write_cmd(self.FRMCTR3, 0x01, 0x2C, 0x2D, 0x01, 0x2C, 0x2D)

# Display inversion control
self.write_cmd(self.INVCTR, 0x07)

# Power control 1
self.write_cmd(self.PWCTR1, 0xA2, 0x02, 0x84)

# Power control 2
self.write_cmd(self.PWCTR2, 0xC5)

# Power control 3 (no matching constant)
self.write_cmd(0xC2, 0x0A, 0x00)

# Power control 4 (no matching constant)
self.write_cmd(0xC3, 0x8A, 0x2A)

# Power control 5 (no matching constant)
self.write_cmd(0xC4, 0x8A, 0xEE)

# VCOM control 1
self.write_cmd(self.VMCTR1, 0x0E)

# Display inversion off
self.write_cmd(self.INVOFF)

# Memory access control
self.write_cmd(self.MADCTL, 0x18)

# Pixel format set
self.write_cmd(self.PIXFMT, 0x05)

# Positive gamma correction
self.write_cmd(self.GMCTRP1, 0x02, 0x1C, 0x07, 0x12, 0x37, 0x32, 0x29, 0x2D, 0x29, 0x25, 0x2B, 0x39, 0x00, 0x01, 0x03, 0x10)

# Negative gamma correction
self.write_cmd(self.GMCTRN1, 0x03, 0x1D, 0x07, 0x06, 0x2E, 0x2C, 0x29, 0x2D, 0x2E, 0x2E, 0x37, 0x3F, 0x00, 0x00, 0x02, 0x10)

# Normal display mode on
self.write_cmd(self.NORON)

# Display on
self.write_cmd(self.DISPLAY_ON, 0x64)

```

rdagger, 2024-06-24:
Great job figuring it out!

closed 2024-06-24 by rdagger