rdagger/micropython-ili9341 #33: Incorrect color in draw_text8x8

QuirkyCort, 2024-07-01:
The colors in draw_text8x8 are incorrect. Example...

```
display.clear(color565(0, 255, 0))
display.draw_text8x8(0, 0, "foo    bar",  color565(0, 0, 0), background=color565(0, 255, 0))
```

The background color of the text will differ from the rest of the background.

This occurs because draw_text8x8 uses the Framebuffer to draw the text, and the endianness of the FrameBuffer differs from that of the ILI9341. The current code attempts to work around this by extracting the RGB values from the color, then recreating it in BRG order using color565(). However, this does not result in correct colors.

A better solution would be to create the color with bytes order swapped.

In ili9341.py, replace lines 630 - 631...

```
            fbuf.fill(color565(bg_b, bg_r, bg_g))
        fbuf.text(text, 0, 0, color565(b, r, g))
```

...with...

```
            bg_color = (bg_r & 0xf8) | bg_g >> 5 | (bg_b & 0xf8) << 5 | (bg_g & 0x1c) << 11
            fbuf.fill(bg_color)
        fb_color = (r & 0xf8) | g >> 5 | (b & 0xf8) << 5 | (g & 0x1c) << 11
        fbuf.text(text, 0, 0, fb_color)
```

Tested in on my device and it appears to provide correct colors now.

rdagger, 2024-07-01:
Thanks for catching the bug and providing a detailed solution!  I corrected it as follows:
```
        # Swap background color bytes to correct for framebuf endianness
        b_color = ((background & 0xFF) << 8) | ((background & 0xFF00) >> 8)
        fbuf.fill(b_color)
# Swap text color bytes to correct for framebuf endianness
t_color = ((color & 0xFF) << 8) | ((color & 0xFF00) >> 8)
fbuf.text(text, 0, 0, t_color)
```

I also included a new demo called _demo_fonts8x8_bgcolor.py_ to test it.

QuirkyCort, 2024-07-02:
Right. That was silly of me. I could have just swapped the bytes and avoided extracting the RGB...

closed 2024-07-02 by QuirkyCort