rdagger/micropython-ili9341 #36: show() function

Rickroller01, 2024-10-12:
Is there a way to implement manually updating the display similar to the show() function in the ssd1306 library. My guess is adding another buffer to the Block() function.

rdagger, 2024-10-12:
This issue is similar to https://github.com/rdagger/micropython-ili9341/issues/35#issue-2568191496

Rickroller01, 2024-10-13:
> This issue is similar to [#35 (comment)](https://github.com/rdagger/micropython-ili9341/issues/35#issue-2568191496)

I have noticed that. Will I be able to replace the block function with an extra framebuffer then add a show() function that is basically the block function?

rdagger, 2024-10-13:
Improving the library properly will take some work. A simple approach would be to modify the block function to draw to a framebuffer and then add a show method, but this wouldn't fully take advantage of the performance benefits offered by the framebuffer class.

In MicroPython v1.20.0, several useful framebuffer methods were added, such as _ellipse_, _poly_, and _rect_ (with a new fill option). These methods are implemented in C, making them significantly faster and more efficient than their Python equivalents.

To improve the library properly, a secondary set of shape-drawing methods should be added to fully leverage these native framebuffer methods. This would provide the best performance and functionality by taking advantage of the underlying C implementations.

However, this functionality is already achievable by directly creating a framebuffer, drawing on it using the native methods, and then using the _display.draw_sprite_ method to transfer the framebuffer to the display.

closed 2024-12-16 by rdagger