rdagger/micropython-ili9341 #39: Image in wrong place on 128x128 display

StuMunsonMcGee, 2025-02-07:
I'm using the ILI9341 driver and testing it with two displays (only one at a time).  It works just fine after some tweaking for the larger display (128x160) but with the smaller display (128x128) it seems that the screen is "shifted" up and to the left.  At the bottom and right edges are "confetti" stripes.  Here is the code I use to initialize the display

screen = "Small"
    
    if screen[0].upper() == "S":
        screen_w = 128
        screen_h = 128
        screen_bgr = False
    
    elif screen[0].upper == "M":
        screen_w = 160
        screen_h = 128
        screen_bgr = True
    
    
    spi = SPI(1, baudrate=40000000, sck=Pin(10), mosi=Pin(11), miso=None)
    display = Display(spi, dc=Pin(16), cs=Pin(18), rst=Pin(17), mirror=True,
                      width=screen_w, height=screen_h, rotation=90, bgr=screen_bgr)

Any info on how to shift the image down (maybe a pixel) and right (also about a pixel) would be greatly appreciated.  Thanks,

Stu

rdagger, 2025-02-07:
Please send a link to the board that is not working.

StuMunsonMcGee, 2025-02-08:
https://www.aliexpress.us/item/3256807160333635.html?srcSns=sns_SMS&spreadType=socialShare&bizType=ProductDetail&social_params=21748081669&aff_fcid=ee459a3053e74f28813e0c32353e6fc3-1739038763307-03706-_mMOkErt&tt=MG&aff_fsk=_mMOkErt&aff_platform=default&sk=_mMOkErt&aff_trace_key=ee459a3053e74f28813e0c32353e6fc3-1739038763307-03706-_mMOkErt&shareId=21748081669&businessType=ProductDetail&platform=AE&terminal_id=e4ef2447797d4bca93ca5369b7f39f62&afSmartRedirect=y&gatewayAdapt=glo2usa4itemAdapt

its the small one.

Also, I have the same pair (126x128 and 128x128) with the ST7735 driver and they have the same issue.  The 160x128 works fine, the smaller one seems as though the output is shifted up and to the left a pixel or two.

[image]
The photo shows the problem.

rdagger, 2025-02-08:
According to the link you provided above, your 1.44-inch display uses an ST7735 instead of an ILI9341.  The displays are similar but are not entirely compatible.  Please take a look at this [issue](https://github.com/rdagger/micropython-ili9341/issues/32#issuecomment-2186710219).  I provide the initialization and @loreblb provides additional fixes to make it work.

StuMunsonMcGee, 2025-02-09:
Thanks for your help!  I by accident had another new, unopened display of the same size.  The label clearly says it has an "ILI9341 Driver Interface SPI for Arduino Color 1.44inch TFT". When I used it, I still have the "stripes" on the right and bottom edges.  The offset command did work for the ST7735 drivers that I have.  Is there a similar command for your ILI9341 driver?  Or should I try to add it into the code as in the example you linked?

rdagger, 2025-02-09:
I would try the code in the linked example.

rdagger, 2025-02-24:
Even if two displays use the same ILI9341 driver, their panel quality can vary (e.g., different glass, filters, and wiring). This can lead to differences in color richness, brightness, or bleeding when running identical code.

### Potential Fixes

- Lower the SPI Speed:  
-- Try reducing the SPI clock. High speeds can cause signal integrity issues, leading to ghosting or color artifacts.
- Tweak or Disable Custom Gamma
-- Disable Custom Gamma: Pass gamma=False when initializing the display. This will use the built-in default gamma correction values rather than the default ones that might not suit your specific panel.
-- Adjust the Gamma curve: If disabling gamma doesn’t help, you can experiment with different gamma curve presets. For instance, in the code that writes to GAMMASET (0x26), you might change the default 0x01 to another built-in value like 0x04 or 0x08.
--You can also edit the positive (GMCTRP1) and negative (GMCTRN1) gamma arrays to fine-tune color balance, but this is more advanced trial-and-error.

Here's a pic of the color wheel demo on a 320x240 ILI9341 display with a low angle step size:

[image]

rdagger, 2025-03-02:
I added support for the ST7355.  
See [demo_st7735s.py](https://github.com/rdagger/micropython-ili9341/blob/master/demo_st7735s.py)

closed 2025-03-02 by rdagger