rdagger/micropython-ili9341 #40: [Question] Is it compatible with MSP3520 and 480x320 RGB?

b24fun, 2025-04-26:
I want to buy a screen but idk if its compatible with this library. Can someone help me?

[image]
https://www.aliexpress.com/item/32847628219.html#nav-description

rdagger, 2025-04-26:
This library is for the ILI9341, and it is also compatible with the ST7735.  The display you posted is an ILI9488.  I have not tried, but I don't think it would be compatible without modifications.

b24fun, 2025-04-26:
Question. I look up on amazon and i saw your review, is that the display that you use to make this library?

> This library is for the ILI9341, and it is also compatible with the ST7735. The display you posted is an ILI9488. I have not tried, but I don't think it would be compatible without modifications.

b24fun, 2025-04-26:
And what about this one?
Sadly i cant buy from amazon because the delivery costs a whooping $25
[image]

rdagger, 2025-04-26:
The 2.8" HiLetgo ILI9341 which is available on Amazon was the display I used to create the library.  Last year I purchased a 2.4" ILI9341 on Aliexpress from Tenstar Robot and it works fine and is relatively inexpensive.

closed 2025-04-30 by rdagger